zynthian/zyncoder #3: Pitch Bend NoteOff

mheidt, 2017-10-03:
When the note off event is triggered while the pitch bend wheel is used, the message is not processed.
You can here the note-off of the initial pitch.
Furthermore when you NoteOn pitched, it is not processed either.

In the following example I press c, bend down and release the key while the pitch is still bent.
Then I press the key again and pitch to normal.

MIDI EVT: 144,48,125
MIDI EVT: 224,85,53
MIDI EVT: 224,123,30
MIDI EVT: 224,41,10
MIDI EVT: 224,0,0
MIDI EVT: 144,48,0
MIDI EVT: 224,24,6
MIDI EVT: 224,74,18
MIDI EVT: 224,20,37
MIDI EVT: 224,85,53
MIDI EVT: 224,127,63
MIDI EVT: 224,0,64
MIDI EVT: 224,127,63
MIDI EVT: 224,52,45
MIDI EVT: 224,90,22
MIDI EVT: 224,24,6
MIDI EVT: 224,0,0
MIDI EVT: 144,48,125
MIDI EVT: 224,41,10
MIDI EVT: 224,3,33
MIDI EVT: 224,77,51
MIDI EVT: 224,118,61
MIDI EVT: 224,0,64
MIDI EVT: 144,48,0

@jofemodo  
Do you have time to check this or should I take a look myself?

jofemodo, 2017-10-04:
Hi @mheidt !
Are you using the global fine tuning feature? (different to 440)

mheidt, 2017-10-04:
Yes, 442

jofemodo, 2017-10-11:
Can you try again with the last changes to the zyncoder MIDI router?

mheidt, 2017-10-13:
I just installed a complete new zynthian. New image, zynthian-update. Even twice. But I think, it is not necessary anymore, right?
I still have the same behaviour. And I changed to 442 already.

mheidt, 2017-10-31:
The issue is, that the pitchbend event is only taken into account, when note is on.
If you pitch bend while note-off, it is not stored and used for the next note-on.
where would you store it!?

mheidt, 2017-11-09:
If you tell me, where I can store the pitchbend values while beint note-off, I could give it a try.

jofemodo, 2017-11-10:
Hi @mheidt !

+ Please, check if the problem is present when no tuning adjustment is done (440Hz). I've tested and it works OK for me.

+ In the zyncoder.c file, line 552, there is a TODO note =>  "modify pitch on pitchbending events!"

I think the problem is that when fine tuning is enabled, a PB message is inserted after every note-on message. This inserted PB value is calculated using the fine-tuning frequency, but it doesn't have into account the current (last) PB value received from the external PB wheel...

The PB value inserted after a note-on should be corrected with this current/last PB value received.

Currently the last value for every channel & CC is stored in an array, but not for PB (it's not a CC). This should be added and then the correction should be trivial.    

Anyway, i've to make some test to be sure of my analysis being correct ;-)

What do you think?

Regards,

mheidt, 2017-11-10:
@jofemodo,
I tested 440Hz 9d ago. Same false behaviour.
As I said, my experience was, that pitchbends during note-off are swallowed.

jofemodo, 2017-11-10:
Hi @mheidt!

I just pushed a new branch "gh003" that tries to fix the problem. I couldn't test it, because i've no Zynthians here, but perhaps you could give it a try ;-)

Regards

mheidt, 2017-11-11:
The initial issue seems to be solved:
When I send Note-Off (release Key), the pitchbend is considered.

But not the second one. I press a key while the wheel has a non-zero (63) position, the Note-On is not considering this initial (silently sent) pitchbend value.

jofemodo, 2017-11-13:
Hi @mheidt !
I fixed some bugs. Try now, please ;-)

mheidt, 2017-11-13:
Yes..great!
I have to test with a detuned engine yet. (425Hz or the like)

mheidt, 2017-11-15:
When I detune down to 392, this is rock bottom...i can't pitch down :)

jofemodo, 2019-04-06:
Is this solved, @mheidt ?

mheidt, 2019-04-07:
Yes. Perfect

closed 2019-04-07 by mheidt