zynthian/zyncoder #8: Implement MIDI CC "relative modes"

jofemodo, 2019-04-06:
Proposal by uli at forum thread:

https://discourse.zynthian.org/t/relative-changes-on-control-change/2862

-Relative #1:
  Up: Value >64, higher is faster
  Down: Value <64, lower is faster
  Every 2nd value is 64
-Relative #2:
  Up: Value >0, higher is faster
  Down: Value <=127, lower is faster
  Every 2nd value is 0
- Relative #3:
  Up: Vaue >16, higher is faster
  Down: Value <16, lower is faster
  Every 2nd value is 16

MrDham, 2021-03-23:
Hello, 

A few month ago, when we discussed about Fixe of #29, you proposed implement a config flag for disabling the "relative mode" auto-detection code.  

Has this idea made its way ? 
This feature would be nice when using MIDI CC in a strict "absolute mode". 

Regards

jofemodo, 2021-07-05:
Yes. This is implemented from time ago:

https://github.com/zynthian/zyncoder/issues/29

Closing this, as nobody seems to need other relative modes apart from mode 1 ;-)

closed 2021-07-05 by jofemodo