zynthian/zyncoder #10: I2C HWC sometimes stops responding

riban-bw, 2019-04-17:
Sometimes the I2C HWC stops interacting with zynthian. This may be after zynthian has been restarted. Subsequent restarts of zynthian have no effect. Rebooting HWC is required.

riban-bw, 2019-04-17:
This may occur if HWC has detected a controller change whilst zynthian is not running (TBC). Solution may be to clear all I2C registers during lib init. Assign to @riban-bw .

riban-bw, 2019-04-17:
I have a patch for this. I need to write a test case to prove this fixes the actual issue and run against current and fixed implementation then will submit a pull request.

riban-bw, 2019-04-17:
Test case:

	Boot RPi into Zynthian
	SSH to RPi
	Stop zynthian: systemctl stop zynthian
	Adjust rotary encoders and push buttons
	Start zynthian: systemctl start zynthian
	Check rotary encoders and push buttons control zynthian UI

Fails before PR #11. Passes after PR #11. PR #11 clears the I2C HWC registers during intialisation. It may be advantageous if the HWC had a reset function which would mean a single I2C message rather than the iteration performed by this PR. [Feature request](https://github.com/riban-bw/I2C-Hardware-Controller/issues/12) submitted to I2C HWC project.

riban-bw, 2019-04-18:
Now uses I2C reset command. PR #11 fixes this issue. Awaiting peer review.

jofemodo, 2019-04-22:
Code seems correct, but i can't really test it because i've not a proper testing device ;-)

riban-bw, 2019-04-22:
Updated and all working as expected. HWC is reset at zynthian startup and menu dividers are skipped as expected. Cheers!

closed 2019-04-22 by riban-bw