zynthian/zyncoder #15: Change bold / long button press behaviour

riban-bw, 2020-04-08:
The current implementation of short / bold / long button press can be awkward. There is no indication of where you are within the process which can make it difficult to differentiate between the three activities, especially when in a stressful environment, e.g. live performance. I find the long press un-intuitive. I would expect it to trigger after a period without the need to release the button which is common user interface behaviour.

**Describe the solution you'd like**
A visual indication of when the button press state changes:
- NOT PRESSED
- SHORT PRESS
- BOLD PRESS
- LONG PRESS

This could be the colour of the corresponding control changing.

Trigger LONG press activity when long press timer expires, not when button released, i.e. trigger events on:
- SHORT RELEASE
- BOLD RELEASE
- LONG PRESS

**Describe alternatives you've considered**
An indication of LONG PRESS trigger may improve the situation but is not ideal. I would like it if I can just press and hold the button and the activity occurs. I can then release the button when I see or hear the event, e.g. ALL NOTES OFF.

**Additional context**
Current behaviour is driven by button events (button down, button up). Time checks are performed when the button up event is triggered. This feature request may require time based events. This may be possible with Python threading.Timer or by using a check within a program loop. It may be advantageous to offload this to the button controller but I think this may prove challenging with the quantity of interfaces supported, some (most?) of which may not support this. (The I2C Controller could probably provide each state in its event value.)

jofemodo, 2020-04-08:
Hi @zynthianers!

I changed the zyncoder switches behaviour so long pushes will be detected before release, as @riban-bw has proposed. I don't consider adding GUI feedback for bold/short pushes because they are too fast and eyes too slow. But ... i added some config vars for allowing to adjust the timing values:

+ ZYNTHIAN_SWITCH_LONG_US
+ ZYNTHIAN_SWITCH_BOLD_US

It should be added to webconf UI's adjustments ;-)

riban-bw, 2020-04-08:
This is not working for me. I still see old behaviour. Could it be related to my use of I2C_HWC wiring? I see you fixed this in zyncoder.

riban-bw, 2020-04-08:
I2C fixed with push https://github.com/zynthian/zyncoder/commit/53d9536fdece213c239072615e7ec04803d3c9bd. This is on a branch and needs merging. PR submitted: https://github.com/zynthian/zyncoder/pull/16

closed 2020-04-09 by jofemodo