zynthian/zyncoder #20: MIDI jitter

riban-bw, 2020-05-14:
**Describe the bug**
MIDI messages sent through the zynrouter have jitter added approximately equal to the audio latency time.

**To Reproduce**
Steps to reproduce the behaviour:
1. Route JACK MIDI directly from jack-midi-clock to libzynseq
2. Enable debug output within libzynseq
3. Set MIDI clock tempo and start clock.
4. Observe sequencer locks to MIDI clock and clock rate does not change
5. Repeat through zynrouter

**Expected behaviour**
Clock reaches sequencer at regular intervals.

**Expected behaviour**
Clock reaches sequencer at irregular intervals, deviating by multiples of JACK buffer size.

**Setup:**
HARDWARE
Raspberry Pi 4 Model B Rev 1.2
Soundcard: HifiBerry DAC+ light
Display: WaveShare 7 HDMI+USB
Wiring: I2C_HWC
GPIO Expander: Not detected

SYSTEM
Raspbian GNU/Linux 10 (buster)
Build Date: 2020-04-09

SOFTWARE
zyncoder: master (0165831)
zynthian-ui: feature/stepseq (74128db)
zynthian-sys: master (a5eba59)
zynthian-data: master (9cccef5)
zynthian-webconf: master (f04e73b)

**Additional context**
It appears that MIDI messages are simply being stacked into the next process cycle buffer starting at offset zero rather than considering the actual time offset of the incoming MIDI message. Previous implementation of libzynseq used this method but current version uses the MIDI event _time_ parameter to identify its offset from start of process cycle time. The current zynrouter implementation introduces variable jitter to the clock signal which may be imperceptible to many users though it may cause significant impact to others. This will become a significant issue when MIDI controller sequencing is added which will interpolate CC messages in the JACK time domain. In fact it may be an issue now with CC messages being stacked together at start of each process cycle, leading to zipping effect.

I anticipate that a delay of one JACK process (5.6ms at 256 buffer size, 44100 samples per second) may be required to implement a low (~1ms) jitter router. I believe the latency may already be implemented so this should not worsen matters. A fixed latency, low jitter solution is likely to provide better perceived performance than a variable jitter and hence variable latency solution.

jofemodo, 2020-05-14:
Hi @riban-bw !
I fixed it. Update & try now ...

jofemodo, 2020-05-14:
Hi @riban-bw !
Did you test the jitter with the fixed ZynMidiRouter?

riban-bw, 2020-05-17:
I measure click jitter below 1ms which is great thanks.

commit c3623839 (2020-05-14): Fix #20: Refactorize ZynMidiRouter MIDI event managing code for correctly use frame time and avoid Jitter.
closed 2020-05-17 by riban-bw