zynthian/zyncoder #26: add MIDI Filter/Routing based on Port

orcascc, 2020-10-23:
Hy,

please add the possibility to filter/route midi based an the port the data came from.

as an example: I have two Keyboards, both sending on MIDI Channel 1, but they are connected via extra USB-MIDI cables (or direct via USB). With the port based filter/router I could change the MIDI Channel from a keyboard to another channel or filter data from a Keyboard.

Thanks, Andy

riban-bw, 2020-10-24:
The OP has keyboards that cannot have their MIDI channel changed unfortunately.

Such a feature would need to enumerate all USB MIDI devices and allow rules for each. This enumeration may change depending on various factors resulting in different results. It is possible to enumerate using USB hub and port indices which probably offers the most consistent user experience in this case where the same keyboard tour is being used though some users may expect unique keyboard types to always behave the same no matter where they are plugged!

This is a niche requirement that requires some consideration and effort so is likely to be a low priority deliverable, i.e. someone is going to want the challenge 😀 ... and have the time.

Pastitas, 2020-12-03:
This feature could also allow for MPE device filtering, so i will try to find time to look into it

riban-bw, 2024-03-12:
Device based MIDI routing is now implemented.

closed 2024-03-12 by riban-bw