zynthian/zyncoder #28: Midifilter rules: Add the possibility to filter system realtime events (clock, start, continue, stop, sensing, reset)

mark-orion, 2020-11-12:
**Is your feature request related to a problem? Please describe.**
I do not want clock events to go through the filter and create "clock mayhem" in the form of unwanted loops, double pulses etc. In other situations I want to filter start,stop,continue selectively so that external sequencers do not start. I might even want to map "start" to "continue" so that I can arm an external sequencer for recording and then trigger the recording with "continue".
I have solved all these problems with mididings in the past and would love to see the same flexibility in Zynthians MIDI filter.

**Describe the solution you'd like**
Filtering options for system realtime events. They could be similar to mididings:
IGNORE SYSRT filters all realtime events.
IGNORE SYSRT_CLOCK no clock passthru
....

**Describe alternatives you've considered**
Use mididings as an alternative, but would prefer a "native" Zynthian solution that makes use of the excellent web configuration.

jofemodo, 2020-11-12:
Did you note the "[x] System Messages" on the UI's admin menu? Perhaps it solves partially what you propose ...

riban-bw, 2024-03-12:
Closing this ticket because the OP has not responded to the suggested resolution. Assuming current system message filter is appropriate.

commit e1beb97d (2020-11-12): Improves #8, Fixes #28: Don't confuse repeated values of 64 with relative mode.
closed 2024-03-12 by riban-bw