zynthian/zyncoder #29: MIDI Controle Change - Consecutive messages with "value=64" are skipped

MrDham, 2020-11-12:
**Describe the bug**
When we send Control Change messages to the MIDI input of the Zynthian, all messages with "Value != 64" are received, but if "Value == 64" then redundant message are skipped. 

Remark: 
Usually it is not a problem except if we combine two CC# (e.g. CC#16 MSB and CC#48 LSB) in order to have 14 bits resolution. 
In such condition we send LSB first and receiver waits for the MSB to take into account both [MSB and LSB] in synchronisation in order to avoid "jittering". (Doing the opposit is not possible if we want to keep compatibility with 7 bits resolution). 
In the case of "MSB Value == 64", MSB just come once, other LSBs are not taken into account and we loose 14 bit resolution arround MSB=64.   

**To Reproduce**
Steps to reproduce the behavior:
Open MIDI monitor in webconf
1. send consecutive CC#16 messages with Value = 63
2. send consecutive CC#16 messages with Value = 64
3. send consecutive CC#16 messages with Value = 65

**Expected behavior**
1. All CC#16 (value=63) are received
CH#01 CONTROL_CHANGE 16 => 63
CH#01 CONTROL_CHANGE 16 => 63
CH#01 CONTROL_CHANGE 16 => 63
...

2. All CC#16 (value=64) are received
CH#01 CONTROL_CHANGE 16 => 64
CH#01 CONTROL_CHANGE 16 => 64
CH#01 CONTROL_CHANGE 16 => 64
...

3. All CC#16 (value=65) are received
CH#01 CONTROL_CHANGE 16 => 65
CH#01 CONTROL_CHANGE 16 => 65
CH#01 CONTROL_CHANGE 16 => 65
...

**observed behavior**
1. All CC#16 (value=63) are received **OK**
CH#01 CONTROL_CHANGE 16 => 63
CH#01 CONTROL_CHANGE 16 => 63
CH#01 CONTROL_CHANGE 16 => 63
...

2. Just one CC#16 (value=64) is received **NOK**
CH#01 CONTROL_CHANGE 16 => 64

3. All CC#16 (value=65) are received **OK**
CH#01 CONTROL_CHANGE 16 => 65
CH#01 CONTROL_CHANGE 16 => 65
CH#01 CONTROL_CHANGE 16 => 65
...

**MIDI logs observed in a 14 bit application case**
// In following sequence, CC#48 and CC#16 (value=63) are received alternativelly
CH#01 CONTROL_CHANGE 48 => 117
CH#01 CONTROL_CHANGE 16 => 63
CH#01 CONTROL_CHANGE 48 => 126
CH#01 CONTROL_CHANGE 16 => 63

// In following sequence, several CC#48 are received for just one CC#16 (value=64)
CH#01 CONTROL_CHANGE 48 => 2
CH#01 CONTROL_CHANGE 16 => 64
**CH#01 CONTROL_CHANGE 48 => 8
CH#01 CONTROL_CHANGE 48 => 17
CH#01 CONTROL_CHANGE 48 => 27
CH#01 CONTROL_CHANGE 48 => 40
CH#01 CONTROL_CHANGE 48 => 62
CH#01 CONTROL_CHANGE 48 => 74
CH#01 CONTROL_CHANGE 48 => 96
CH#01 CONTROL_CHANGE 48 => 114 All these messages in bold cannot be taken into account because MSB doesn't arrive**

CH#01 CONTROL_CHANGE 48 => 1
CH#01 CONTROL_CHANGE 16 => 65

**Setup:**
 - SD-image
Raspbian GNU/Linux 10 (buster)
Build Date: 2020-09-05

 - Hardware Config
Raspberry Pi 3 Model B Plus Rev 1.3
Soundcard: HifiBerry DAC+
Display: PiScreen 3.5 (v2)
Wiring: MCP23017_EXTRA
GPIO Expander: MCP23017

 - Software Version
zyncoder: master (0d4f15c)
zynthian-ui: master (2ea9647)
zynthian-sys: master (a06533e)
zynthian-data: master (ee9a229)
zynthian-webconf: master (d25055c)

**Additional context**
N/A

jofemodo, 2020-11-12:
I think i have a solution. Please, update and test.

Thanks!

closed 2020-11-12 by jofemodo

MrDham, 2020-11-12:
Hi, @jofemodo, 

It is much better, but still remain some transitory effect when leaving state where the cc value = 64. 
it seems that there is some "overshot" of cc value. 

**- Decreasing from CC value = 64 to 63:** 

CH#01 CONTROL_CHANGE 48 => 8
CH#01 CONTROL_CHANGE 16 => 64
CH#01 CONTROL_CHANGE 48 => 5
CH#01 CONTROL_CHANGE 16 => 64
CH#01 CONTROL_CHANGE 48 => 2
CH#01 CONTROL_CHANGE 16 => 64
CH#01 CONTROL_CHANGE 48 => 126
CH#01 CONTROL_CHANGE 16 => 63
CH#01 CONTROL_CHANGE 48 => 113
**CH#01 CONTROL_CHANGE 16 => 62 -> It seems this "62" should rather be a "63"**
CH#01 CONTROL_CHANGE 48 => 107
CH#01 CONTROL_CHANGE 16 => 63
CH#01 CONTROL_CHANGE 48 => 110
CH#01 CONTROL_CHANGE 16 => 63
CH#01 CONTROL_CHANGE 48 => 94
CH#01 CONTROL_CHANGE 16 => 63

**- Increasing from CC value = 64 to 65 (and staying at the limit between 64 and 65):** 

CH#01 CONTROL_CHANGE 48 => 111
CH#01 CONTROL_CHANGE 16 => 64
CH#01 CONTROL_CHANGE 48 => 123
CH#01 CONTROL_CHANGE 16 => 64
**CH#01 CONTROL_CHANGE 48 => 120
CH#01 CONTROL_CHANGE 16 => 64
CH#01 CONTROL_CHANGE 48 => 126
CH#01 CONTROL_CHANGE 16 => 64
CH#01 CONTROL_CHANGE 48 => 7
CH#01 CONTROL_CHANGE 16 => 65
CH#01 CONTROL_CHANGE 48 => 1
CH#01 CONTROL_CHANGE 16 => 66-> It seems this "66" should rather be a "65"
CH#01 CONTROL_CHANGE 48 => 126
CH#01 CONTROL_CHANGE 16 => 64 -> This is normal we are at the limit betwen 64 and 65
CH#01 CONTROL_CHANGE 48 => 1
CH#01 CONTROL_CHANGE 16 => 65
CH#01 CONTROL_CHANGE 48 => 4
CH#01 CONTROL_CHANGE 16 => 66-> It seems this "66" should rather be a "65"
CH#01 CONTROL_CHANGE 48 => 1
CH#01 CONTROL_CHANGE 16 => 65
CH#01 CONTROL_CHANGE 48 => 4
CH#01 CONTROL_CHANGE 16 => 65-> Stable 65** 
CH#01 CONTROL_CHANGE 48 => 7
CH#01 CONTROL_CHANGE 16 => 65
CH#01 CONTROL_CHANGE 48 => 10
CH#01 CONTROL_CHANGE 16 => 65

jofemodo, 2020-11-12:
The problem is the "relative/absolut mode detection" code. The "relative mode", used by many MIDI controllers with "infinite rotaries" interleaves a value 64 with an increment/decrement value. Zynthian tries to detect the type of controller by tracking the values 64 and smoothing change between modes, so you can connect both types of controllers to zynthian and all of them will behave nicely. The problem is that transition is not totally smooth and some ticks can be lost or confused when passing the 64 level. In your case, it briefly changes to relative mode.

I think the only way of matching 100% the received MIDI data is disabling this code with a config option.

Regards,

MrDham, 2020-11-12:
Hi @jofemodo, 

Ok, I got it. 

I won't complain if you prefer reverting the change: initial behaviour was OK for most users. My use case is rare, it doesn't worth risking some side effect... 

Having "Auto Relative-Mode" enabling configurable could be a nice solution effectivelly.

MrDham, 2020-11-14:
Hi, 

I don't know if it's properly the purpose of this variable but I could fix it for my own needs by changing the initialization line "midi_ctrl_automode=1; " into "midi_ctrl_automode=0; "

Thanks for the explainations above, now it works for me

jofemodo, 2020-11-14:
But if you change this you couldn't update your zynthian ... i leave this open until i implement the config flag for disabling the "relative mode" code.
Regarding the last change i did, it's an improvement to the auto-detection algorithm and shouldn't have side effects but smoothing the mode transitions ... i hope! ;-)

Regards,

MrDham, 2021-04-18:
Hello @jofemodo ,

After some evolution of this topic on my side it appears that this bug was rather invalid : 

- My problem appears just because I send to the Zynthian LSB before MSB, bufferization of LSB allows me to use [MSB and LSB] in synchronisation. But, after some investigation, it appears that it is against the standard: 
    " The LSB value is assumed zero when an MSB arrives.
     The MSB may or may not be followed by an LSB."
 This in an awfull flaw of the MIDI standard because there is no way to synchronize MSB and LSB and have 7bit compatibility at the same time, but the standard it is. 
 (If you are currious, all is explained at https://www.hakenaudio.com/mpe). 
So If I send MSB before LSB, as per standard, the single CC#16 (value=64) is not a problem. 

- As we discussed above, the solution chosen previously [465929c](https://github.com/zynthian/zyncoder/commit/465929c71c3cbd4200f3aae3019f226b5d633ad7) creates some glitches. If the CC is controling a pitch related parameter, these glitches can be heard (Pitch is one of most precise human perception).  So this change trying to fix my own little non-tandard problem creates a bigger problem for all other users. 

**As a conclusion it is probably better to revert the change [465929c](https://github.com/zynthian/zyncoder/commit/465929c71c3cbd4200f3aae3019f226b5d633ad7) and re-close this bug as invalid.** 

Still I am very intesrested in having "Auto Relative-Mode" activation configurable in the webconf ;-)

jofemodo, 2021-04-18:
Commit 465929c is an improvement and, AFAIK, it doesn't have secondary effects. No need for reverting.
But the problem is still there. When playing around the value 64 level, some spurious message can be added by the auto-mode code. I don't think this is could be noted on a real-life case, but if somebody report problems related with this, a flag for disabling this code could be implemented easily. Until then, i prefer to don't add extra complexity for solving a non-existing problem ;-)

Thanks!

MrDham, 2021-04-18:
Thanks @jofemodo for you answer,

In fact, I am facing the problem while controling the frequency of an oscillator with a (14 bit) CC associated to the pitch antenna of a MIDI Theremin (Ok, it is not so common but why not). The glitch resulting from spurious message is audible and is at the center of playing range. That's why I'd like it to be fixed although I totaly understand that I am probably the only one out there who noticed such a problem  ;-)

The patch that I am using on Zynthian is a Pure Data patch: https://github.com/MrDham/Pure-Data-FM-Ribbon-Synth.

jofemodo, 2021-04-20:
Hi @MrDham!

I just added a new option for disabling the CC relative automode. It's on the webconf's MIDI options. Just update and test (testing branch!).

Thanks

MrDham, 2021-04-20:
@jofemodo 
A great thank you! I'll test it ASAP. I'll think I'll need to make with a very polished :fave_with_monocle:  

:-)

MrDham, 2021-04-20:
Just tested. It works nicely.

commit 465929c7 (2020-11-12): Improves #8, Fixes #29: Don't confuse repeated values of 64 with relative mode.