zynthian/zyncoder #32: Zynaptik Switch doesn't work for Program Change

wyleu, 2021-03-12:
**Describe the bug**
Selecting program change in the wiring webconf does not work for Program Change . All the other do work

**To Reproduce**
Steps to reproduce the behavior:
1. Go to webconf wiring
2. Select Zynaptik awitch 5 for zynaptik i/o 
3. reboot
4. All other options produce expected behaviour , Program Change does not

**Expected behavior**
I expect to see a MIDI log message for program change when the button is pressed

**Screenshots**

[image]

jofemodo, 2021-03-12:
As you can see, *ALL* program changes are intercepted by the zynthian UI and you can't see them on the ZynMidiRouter output.  

It was implemented like this because most of the standalone engines listen for PGs, what collides with the UI's bank/preset manager.

This should be improved because it's a limitation for some use-cases, specially when using pure MIDI layers. I will leave this issue open and will try to improve this behaviour:

Program Change messages should be deleted from the ZynMidiRouter output **only** when strictly needed for avoiding collision with UI. That is, only when the output goes to any of these engines:

+ ZynAddSubFX
+ FluidSynth
+ setBfree
+ Aeolus
+ Pianoteq

As ZynMidiRouter is engine-agnostic (It doesn't know about layers or engines), i have to implement a flag that can be enabled/disabled for each output port externally. I've to think a little bit about this ... ;-)

Regards,

jofemodo, 2021-03-13:
Hi @wyleu !

I just committed a patch that solves partially the problem. The improved code *only* drop PC messages from "engine" output ports (Main & CH#XX), allowing the PCs to flow thru the rest of output ports (MIDI-OUT, NET, etc.).

If you select, for instance, "MIDI-OUT" on the MIDI-log selector, you should see the PCs generated by your switches (don't forget to update first ;-).

Regards,

wyleu, 2021-03-13:
[image]

Ok that programmes the buttons as per https://wiki.zynthian.org/index.php/Zynthian_UI_Users_Guide#MIDI_Program_Change_.26_ZS3_.28SubSnapShots.29

Time to close :-D

closed 2021-03-13 by wyleu

jofemodo, 2021-03-15:
Hi @wyleu !

I just committed a patch that solves completely the issue. Now "Program Change" messages are only blocked for the engines that really need it: Aeolus, FLuidSYnth, setBfree & ZynAddSubFX

commit 10e88c3f (2021-03-15): Fix #32: Implement "Drop Program Change" flag for ZMOPS, but always allowing internal PCs (coming from ZMOP_FAKE_IN).